lhr-solar/Electronics #16: Determine the location of electronics in the car

chaseblock, 2020-06-04:
Right now, the locations of each board and wire in the car are ill-defined. We need to create a lay-out of the electronics in the vehicle, taking into account all communications between boards, all power requirements, and size constraints.

This requires a good deal of work with the mechanical design team.

This issue will be resolved by the following deliverables:
* A description of the required communication and power lines within the car.
* A fully-defined specification of the locations of the electronics within the car.

chaseblock, 2021-02-14:
What's the status on this issue? I think I'm aware of plans for all of the car's electronics, aside from maybe the mounting points for the motor contactor and contactor control board.

Thiago-de-Sousa, 2021-02-14:
Creepy I hadn't opened github in like a semester, just opened it for a class and immediately get hit with an email for this. I'm assuming this will entail some collab between electronics, the new harness team, and enclosures.

HannahSLee14, 2021-02-14:
I'm hoping that doing all the hardware documentation for the boards will make this issue a lot easier to complete! A part of the documentation is the location of the board and all input/output signals and power lines from each of the boards. Once the documentation is done, we can just get all that information and put it in a single doc.