lhr-solar/Electronics #17: Assess the status of the contactor/precharge boards.

chaseblock, 2020-06-04:
I'm not quite sure where we stand on this, and I would like to know the following:

* Do we have assembled contactor/precharge boards?
* Where are they?
* How well do they work?

The design exists in the ContactorDriver folder in this repository.

chaseblock, 2020-06-04:
If we do not have the boards assembled yet, do we have the PCBs ordered, and do we have the parts to assemble them?

SijWoo, 2020-07-15:
Just noticed this ticket, This was already discussed/confirmed but just to have written confirmation:
- The contactor/precharge boards were ordered and and two boards assembled before spring break/quarantine.
- Dr. Hallock has them at pickle. Should be with the rest of the electrical system in the battery room.
- A couple members from power systems team and I tested them before quarantine and they worked.

chaseblock, 2020-07-15:
Good to hear. I'm closing this issue out if that's the case. We can revisit this when we test the whole system in the Fall.

closed 2020-07-15 by chaseblock