lhr-solar/Electronics #24: RTD boards

chaseblock, 2020-06-04:
We need to assemble and test the RTD boards, if this hasn't been done already.

chaseblock, 2020-06-04:
@HannahSLee14 I forget where we are at with this right now. What's the current status?

dimembermatt, 2020-06-05:
Are we planning on using already made (I assume) BPS software for the RTD sensors on the array? If they don't already exist, is that something I can start on?

HannahSLee14, 2020-06-05:
I think that the RTD PCB and parts have been ordered, but it hasn't been assembled yet! I was planning on having my team assemble it during the fall semester. I don't know what the plans are for the code though.

dimembermatt, 2020-06-05:
Alright, then that's something I can set up for interested members or start working on in the coming weeks. A couple of bits of information I need:
- what is the uC used for the board? (I forgot :3)
- what pins are connected for MUX select signals, input, etc?
- maybe a link here to the schematics will be very helpful.

Also, do the RTD boards send data to the MPPT board or the power board?

HannahSLee14, 2020-06-05:
It uses the Nucleo-L432KC. I don't think there's a MUX on the board- right now, all the sensors are connected to analog pins on the microcontroller. This is the link to the schematic ([https://github.com/lhr-solar/Electronics/tree/master/Array/RTD%20Temp%20Board](url)). I'm not entirely sure where it sends data to! All I know is that it uses CAN to communicate with some other system.

dimembermatt, 2020-06-05:
Looks like the software is good. The only thing left is to tune the rtd boards when they are made, which requires assembly and testing.

chaseblock, 2020-07-02:
@SijWoo where are these boards?

SijWoo, 2020-07-02:
They have been ordered but they have not been delivered yet. JLC said they won’t be delivered until it’s safe again (?...) I think they tried delivering it but UT was closed because of the quarantine so no one was there to receive the package and now JLC is holding onto it.