lhr-solar/Electronics #33: Need to add a GPS unit.

chaseblock, 2020-06-21:
We would like to add a GPS unit to the car. This can be used to position the car for increased analysis and tracking purposes.

Things that need to be done:
1. Determine which GPS module should be used. There should be ample rationale behind this decision.
2. Create a design for the telemetry board that uses a GPS module.
3. Assemble the board.
4. Test the design.

chaseblock, 2020-07-11:
Has there been any progress made on this? Do we have a module in mind?

SijWoo, 2020-08-07:
I've used the [MTK3339](https://www.adafruit.com/product/790) before. The interface is UART and the module would periodically transmit string packets to the MCU which would then be parsed for coordinates, speed, etc. Parsing format is the NMEA 0183 standard. For additional strength, this [external antenna](https://www.adafruit.com/product/960) can be easily attached if the correct connector is added onto the board. No other components are needed on the board to attach an external antenna, just a direct connection between the MTK3339 pin and the connector is needed.