lhr-solar/Electronics #34: Need to add an IMU to the telemetry board.

chaseblock, 2020-06-21:
We would like to add an accelerometer/gyro to the telemetry board. This would allow us to track the speed and orientation of the vehicle. Bonus points if there's a magnetometer, since that would allow us to determine the heading of the vehicle.

Steps that need to be taken:
1. Determine the IMU module that we should use.
2. Design a board that uses the module.
3. Assemble and test the board with the IMU.

This is somewhat related to #33.

chaseblock, 2020-07-11:
Has there been any progress made on this? Any chip/module that is looking like a good choice?

SijWoo, 2020-08-07:
I think the BNO055 is good, uses I2C. The IC has calibration firmware preflashed for the magnetometer so we can get precise heading without any matrix calculations if this is one of the data points that telemetry needs