lhr-solar/Electronics #38: Discussion: Should we separate the various boards that we are making?

chaseblock, 2020-07-11:
We make quite a few PCBs for various purposes, and keeping them all in one repository kind of clutters everything. What do people think about starting to separate the various boards into their own repos? I've been coming back to this every once and a while, but I don't really know what I think about it right now.

A few things to consider to start us off:

Pros
====
 - This would eliminate the worry of merging various projects when things are being worked on concurrently.
 - We would not have to download all the electronics projects when wanting to edit just one.
 - Issues would be easier to sort through, since only the relevant ones would be shown for a given repo.
 - Having one repo per project feels like a more logical organization, given how our software is organized similarly.

Cons
====
 - We would have to use a different system for sharing common symbols between projects, while right now there is simply a folder labelled "common" that we use.
 - GitHub requires that issues be linked to a repo (at least I think; someone correct me if I'm wrong), so opening issues like "We need to find a ______ chip that does _____ for this new project we haven't yet started" would be more difficult, since they wouldn't have an obvious home.
 - This would require some effort.

I'm interested to hear what others think on this.

chaseblock, 2020-07-11:
@HannahSLee14 @SijWoo @ClarkPoon @JimothyGreene @dimembermatt

chaseblock, 2020-07-11:
Additional pro: We can integrate GitHub's teams feature to place each repository under the clear ownership of it's proper team, which might eliminate some confusion regarding who is responsible for creating which boards.

SijWoo, 2020-07-13:
I like the idea of separating all the subsystem boards. If we had a main Electronics repo and in that repo we have submodules that links to the subsystems' hardware repos, would the second con be solved? Issues that are general or does not have an obvious home can go to that umbrella repo.
Were you planning to separate each individual board to their own repos (e.g. BPS leader board and BPS minion board in their own repos) or each subsystem has their own hardware repo with all their boards in there?

I am down to have any type of reorganization of this repo, when I made this repo, I sort of did not expect all the systems to use git because the previous subsystem leads at the time were all independent and would not use git... This repo was not that well planned out on my part so the organization of all the boards was flawed since the beginning

chaseblock, 2020-07-23:
I realize that I have not been updating the discussion here.

I think that for the most part we have agreement that we should separate this repo's components into several individual repositories (someone please correct me if I'm wrong; this is based on the conversations that I've had recently).

As far as the organization of the boards goes, I would personally be in favor of separating each KiCAD project into its own repository. I think this would make things like reverting to older revisions of a board easier, since it would no longer revert all changes to every board in that system as well. If anyone feels like the other option would be better, please speak up.

When we want to make the change, I suggest scheduling a time for it to take place so that all members can have any changes that they are working on checked in at that point. We can then separate the projects into several repositories. Hopefully if all goes well, any active development can rebase off of the modified master branch of its proper repository.

How does that plan sound?

closed 2020-09-12 by chaseblock