lhr-solar/Electronics #40: Version number system

SijWoo, 2020-07-30:
On every BPS board we add **vX.Y** to the silkscreen to indicate the version. The X number originally indicated a new design/architecture. The Y number originally indicated subversions which tiny errors were fixed (i.e. forgot to connect a pin, connections are flipped, or a different component with the same functionality is used). All boards with the same X version number are compatible with each other i.e. v2.1 leader board can be connected to a v2.0 minion board. This was the original logic behind the numbers. The BPS is currently on v2.Y. The differences between v1.Y and v2.Y are the following:
**v1.Y:**
- Separate Current, Voltage, and Temperature boards
- Voltage board used LTC6811, Temperature board used LTC2983
- Temperature boards not daisy chained
- Voltage boards are daisy chained

**v2.Y:**
- Voltage and Temperature sensing are combined into one (minion) board
- LTC6811 measures both voltage and temperature
- Minion boards are daisy chained

The newest leader board has the exact same architecture as v2.Y but it interfaces with a new current sensing board (v1.Y and v2.Y use hall effect sensors with an analog output whereas the new leader board uses a shunt resistor with the AS8510 with a SPI interface). The newest leader board also has fan control, bluetooth, and the LEDs/strobe light are on different pins. Everything else that wasn't listed stayed the exact same on the newest leader board (i.e. the voltage/temperature board interface, USB, CAN, EEPROM, Contactor connections/code are still the same).

**Should we have the new leader board labeled as v3.0 or v2.1?** The minion boards (voltage/temperature sensing board) are on v2.Y since only boards that are on v2.Y are allowed to be connected with each other. Even though the minion boards are meant for v2.Y leader boards, they can still be used for v3.0 if we decide on to make the new leader board start on v3.0. If we decide on numbering the new leader board to be v2.1, then it goes against the original intent that the Y number represents tiny errors that were fixed. This is what I'm torn about because I don't want the version numbering system to get confusing... I think going with v2.1 would be the better option, even if the fan board/new current board is not able to connect with the v2.0 leader board.

What do you all think is best?

SijWoo, 2020-07-30:
Btw other boards that Hannah and I have made have been using this version numbering system (e.g. RTD board, Contactor board, telemetry), but I think we should standardize a board version numbering system for the whole electrical team. Any thoughts on this board numbering system? If there are a lot of changes/additions you want to discuss with the numbering system, then we can move this to a new issue ticket
@chaseblock @ClarkPoon @HannahSLee14 @JimothyGreene @dimembermatt @cmontminy

HannahSLee14, 2020-08-02:
I think it's a good idea to standardize the version numbering system for the entire electrical system, and I think the current version numbering system that you have right now is good! 
For the leader board version number, I can see why you are torn about it because using v2.1 suggests that all v2.Y boards are compatible with each other when different v2.Y leader boards are compatible with only certain boards. Using v3.0 suggests that v3.Y boards are only compatible with other v3.Y boards when it is also compatible with some v2.0 boards. However, I agree that going with v2.1 is probably better because the minion boards that we are using (which have v2.0 on them) have already been manufactured, so it makes sense to keep it consistent with that. Changing it to v3.0 means that if we want everything to be consistent, we would have to remanufacture the minion boards.

chaseblock, 2020-08-02:
Since we're changing the interface to one of the measurement boards (current), which I assume was numbered v2.x, by the rules of this system I think the new leader and the new current board should be numbered v3.0.

In addition, synchronizing board numbers seemed like a good idea at one point, but I'm sure that some of the boards will have to change more often than others, and we certainly aren't going to re-manufacture them just to put on a new silkscreen. I propose that we take one of these two options:

1. Stop coordinating numbers between boards, and instead maintain a list of version compatibilities for each board. Maybe a file in the root of the repository that declares which versions of each connecting board the design is compatible with. This would allow us to change version numbers on a board when it significantly changed, rather than when a different board significantly changed. For example, maybe the version file would just be a list formatted something like this: 
```
# Dependencies for BPS-Leader v3.0/v2.1

BPS-Voltage-Board = 2.0
BPS-Current-Board = 3.0
```
2. Put a sticker that says v3.0 on the voltage boards.

HannahSLee14, 2020-08-02:
That's a good point! I didn't really think about that. I'm leaning towards doing the first option of the two. It would be really nice to keep all the version numbers consistent and coordinated, but this way is probably the easiest way to handle the version numbering system, especially when we don't know how often we have to redesign the boards. I think the second option could work, but I think using stickers/labels might be a hassle, particularly when you might have to put stickers on top of other stickers.

SijWoo, 2020-08-04:
I like the first proposal (rip the synchronization). If you're all good with Chase's first proposal then lets make the versioning system the following:

vX.Y which X signifies a new design/feature has been added to the board and Y signifies minor fixes such as trace fixes, pin connection fixes, component rearrangement, etc. Each board's version number is independent of any other board. All dependencies must be documented in the VERSION.md file of each board's repo if there exists any. Inside VERSION.md, it should be formatted like the following:
```
# Dependencies for BPS-Leader v3.0

BPS-MinionPCB = 2.0
BPS-AmperesPCB = 3.0
...
```

Let me know if there's anything that can be improved/changed or if it's good. If there aren't any more suggestions or comments by Friday night 8/7/2020, I'm going to assume this is good so we can make sure the boards are edited with the correction version number before sending the boards to Dr. Hallock early next week.

SijWoo, 2020-08-04:
I'll make sure this is documented in the Electronic Repo's README at the end of the week (or after approval). I'll also update this README to have more useful information.

ClarkPoon, 2020-08-04:
I agree that the first proposal is better. In the second proposal, you would end up where version 2.x and version 3.0 of some boards would be identical, which I think would be confusing.

closed 2020-08-15 by HannahSLee14