lhr-solar/Electronics #44: Licensing

chaseblock, 2020-09-12:
Starting this issue here for lack of a better spot. Right now, we do not have any licenses (as far as I know) explicitly assigned to any of our projects. We should probably do this so that we make it clear under what terms our code/designs can be used. I'm partial to the MIT license (see [here](https://en.wikipedia.org/wiki/MIT_License)). If anyone else has input on this, I'd like to hear it.

SijWoo, 2020-09-15:
I think this is a good idea. I like the MIT license

chaseblock, 2020-09-29:
What do others think? This is what I would be attaching to all repositories:

```
Copyright (c) 2020 UT Longhorn Racing Solar

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
```

@ClarkPoon @cmontminy @dimembermatt @Thiago-de-Sousa @JimothyGreene @HannahSLee14

Thiago-de-Sousa, 2020-09-29:
I'm good with it. Open source gang